spacelab-ufsc/eps2 #1: Hardware: Review the components parameters

andrempmattos, 2020-05-06:
Review the components parameters (check the partnumbers, comments, values and etc).

Tip: Enumerate here the components designators with inconsistency.

YanAzer, 2020-05-09:
**RTD1, RTD2 and P8**: Wrong 3D models, 3D models are not right angle but vertical (partnumber: 0532610871).
**R_ESD**: Wrong comment, ESD protection resistor is not 1M ohms, but 100K ohms or 0.1M ohms (partnumber: AC0805FR-13100KL).
**R87 to R91**: Wrong description and design item ID, SMD Jumper resistors have 1/8W rated power dissipation and not 1/16W specified in comments section (partnumber: CRCW08050000Z0EA).

Finished EPS_INTERFACE schematic sheet.

YanAzer, 2020-05-10:
**J_V5**: Wrong description and design item ID, SMD Jumper resistors have 1/8W rated power dissipation and not 1/16W specified in comments section (partnumber: CRCW08050000Z0EA).
**L7, L8 and L9**: Wrong description, the inductors are from HC9 series and not DR127 series.
**C79 to C81, C117, C118**: Wrong design item ID, the correct should be something like: Cap_Cer_X7R_22uF_25V_1210.

Finished 1_Measurement_and_MPPT shematic sheet.

YanAzer, 2020-05-10:
**R85**: Missing “1/10W” to comment, R35 has it.
**R46,R47 and R93**: Wrong description and design item ID, power dissipation should be 3/4W and not 1/2W (https://www.digikey.com/product-detail/en/vishay-dale/CRCW12060000Z0EAHP/541-0-0UBCT-ND/2222920)
R33, R42 and R44: Wrong description and design item ID, power dissipation should be 1/10W and not 1/8W (https://www.digikey.com/product-detail/en/vishay-dale/CRCW0603150RFKEAC/541-4023-1-ND/7928706)

 Finished 2_Battery_Monitoring_And_ Switches schematic sheet.

YanAzer, 2020-05-10:
Inside 3_MCU_EPS schematic sheet:

**J_V3 and J_V4**: Wrong description and design item ID, SMD Jumper resistors have 1/8W rated power dissipation and not 1/16W (https://www.digikey.com/product-detail/en/vishay-dale/CRCW08050000Z0EA/541-0-0ACT-ND/1180267).
**R65 and R66**: Wrong description and design item ID, SMD Jumper resistors have 0.15W rated power dissipation and not 1/8W = 0.125W (https://www.digikey.com/product-detail/en/vishay-beyschlag/MCT0603MC1002FP500/MCT0603-10-0K-MFCT-ND/3883103).
**R62**: Wrong description and design item ID, SMD Jumper resistors have 0.15W rated power dissipation and not 1/8W = 0.125W (https://www.digikey.com/product-detail/en/vishay-beyschlag/MCT0603MD4702DP500/MCT0603-47K-MDCT-ND/2092130).
**P11**: **SUGESTION** No relevant description text, suggestion: “SWITCH TACTILE SPST-NO 0.05A 12V”, maybe this doesn’t even matter anyway (https://www.digikey.com/product-detail/en/omron-electronics-inc-emc-div/B3U-1000P/SW1020CT-ND/1534357).

YanAzer, 2020-05-10:
Inside 4_Battery_Heater_Controller schematic sheet:

**J_V1, J_V6, R28 and R32**: Wrong description and design item ID, SMD Jumper resistors have 1/8W rated power dissipation and not 1/16W (https://www.digikey.com/product-detail/en/vishay-dale/CRCW08050000Z0EA/541-0-0ACT-ND/1180267).
**C126 and C127**: Wrong value of capacitance in comment, description and design item ID, **these components are also obsolete** (https://www.digikey.com/product-detail/en/murata-electronics/GRM216R71H103KA01D/490-1664-1-ND/587472).
**C36**: Wrong value of rated voltage in comment, description and design item ID. The right value is 10V (https://www.digikey.com/product-detail/en/murata-electronics/GRM31CR71A226KE15L/490-5309-1-ND/2039088).

commit 2f093d6c (2020-05-10): hardware: libraries: Updating obsolete and inconsistent components. closes #1 and #2
closed 2020-05-10 by andrempmattos