spacelab-ufsc/eps2 #2: Hardware: Review the components obsolescence

andrempmattos, 2020-05-06:
Review if the components are available in Mouser or Digikey.

YanAzer, 2020-05-10:
No obsolete components found on EPS_INTERFACE schematic sheet.

YanAzer, 2020-05-10:
Inside 1_Measurement_and_MPPT shematic sheet:

**C72 to 78, C83 to C91, C93, C94, C97 to C101, C108 to C116**: Component obsolete https://www.digikey.com/product-detail/en/murata-electronics/GRM188R71H104KA93D/490-1519-1-ND/587854. **Too many components and they are not ordered! Might missed one…**
**C102 to C107**: Component obsolete https://www.digikey.com/product-detail/en/murata-electronics/GRM21BR71E475KA73L/490-9962-1-ND/5026449
**DQ8, DQ9 and DQ10**: https://www.digikey.com/product-detail/en/vishay-siliconix/SI4010DY-T1-GE3/SI4010DY-T1-GE3TR-ND/6834463

YanAzer, 2020-05-10:
Inside 2_Battery_Monitoring_And_ Switches schematic sheet:

**C40, C41 and C119 to C121**:  https://www.digikey.com/product-detail/en/murata-electronics/GRM188R71H104KA93D/490-1519-1-ND/587854
**C37 and C38**: https://www.digikey.com/product-detail/en/murata-electronics/GRM188R71H102KA01D/490-1494-1-ND/587851

YanAzer, 2020-05-10:
Inside 3_MCU_EPS schematic sheet:

**C58, C59, C62, C63, C122, C124 and C133**: https://www.digikey.com/product-detail/en/murata-electronics/GRM188R71H104KA93D/490-1519-1-ND/587854

YanAzer, 2020-05-10:
Inside 4_Battery_Heater_Controller schematic sheet:

**C32, C34, C126 and C127**: https://www.digikey.com/product-detail/en/murata-electronics/GRM216R71H103KA01D/490-1664-1-ND/587472.

closed 2020-05-10 by andrempmattos