spacelab-ufsc/eps2 #3: Hardware: Review the exported bill of materials

andrempmattos, 2020-05-06:


YanAzer, 2020-05-16:
**+X, +Y, +Z, P3, RBF, -X, -Y, -Z**: Designators without problems.
**+X_CSR, +Y_CSR, +Z_CSR, -X_CSR, -Y_CSR, -Z_CSR**: Designators without problems.
**3V3_ANT, 3V3_EPS, 3V3_MCU, 3V3_OBDH, 5_V, 5V_BEACON_LED, 5V_MAIN_RADIO_LED, VBus**: Designators without problems.
**C62,C77,C78**: have the old obsolete partnumber (https://www.digikey.com/product-detail/en/murata-electronics/GRM188R71H104KA93D/490-1519-1-ND/587854) instead of the supposed new one (https://www.digikey.com/product-detail/en/yageo/CC0805KRX7R9BB104/311-1140-1-ND/303050)
**C3, C4, C13, C14, C102, C103, C104, C105, C106, C107, C130**: Designators without problems.
**C6, C7, C8, C16, C17, C18, C30, C48, C131, C132**: Designators without problems.
**C9, C19**: Designators without problems.
**C10, C20**: Designators without problems.
**C21, C26, C32, C34, C44, C52**: Designators without problems.
**C23, C28, C46, C54**: Designators without problems.
**C24, C31, C49, C56**: Does these components have the right footprint? Curretly they are using 1210 package size (https://www.digikey.com/product-detail/en/panasonic-electronic-components/6TPE220MAZB/P16245CT-ND/4146688)
**C33, C35, C60, C61, C64, C123, C125**: Wrong comment, it is 6.3V and not 10V, these issue is better referenced in #1 (https://www.digikey.com/product-detail/en/avx-corporation/TPSR106K006R1000/478-9265-1-ND/4562130)
**C36, C50, C51, C79, C80, C81, C82, C117, C118**: Designators without problems.
**C37, C38**: Designators without problems.
**C39, C65**: Designators without problems.

YanAzer, 2020-05-20:
**C40, C41, C58, C59, C63, C71, C72, C73, C74, C75, C76, C83, C84, C85, C86, C87, C88, C89, C90, C91, C93, C94, C97, C98, C99, C100, C101, C108, C109, C110, C111, C112, C113, C114, C115, C116, C119, C120, C121, C122, C124, C128, C129, C133**: Designators without problems.
**C57**: Designator without problem.
**C66**: Designator without problem.
**C67, C69**: Designators without problems.
**C68, C70**: Designators without problems.
**D1, D2**: Designators without problem.
**D3**: Designator without problem.
**D4, D9, D11, D13, D15**: Designators without problems.
**D5, D6**: Designators without problems.
**D8**: Wrong voltage in description and design item ID, it is 20V and not 23V (https://www.digikey.com/product-detail/en/on-semiconductor/NSVR0320MW2T1G/NSVR0320MW2T1GOSCT-ND/5213359)
**D10**: Designator without problem.
D17, D18, D19, D20, D21, D22: wrong voltage in description and design item ID, it is 20V and not 23V (https://www.digikey.com/product-detail/en/on-semiconductor/NSVR0320MW2T1G/NSVR0320MW2T1GOSCT-ND/5213359)

YanAzer, 2020-05-21:
**FB1**: Wrong current in description and design item ID, should be 4A #1 (https://www.digikey.com/product-detail/en/murata-electronics/BLM21PG300SN1D/490-1052-1-ND/584499).
**FB3, FB4**: Designators without problem.
**J_V1, J_V3, J_V5, J_V6, R28, R32, R87, R88, R89, R90, R91**: Designators without problem.
**J_V4**: Designator without problem.
**L1, L2**: Wrong “DC Resistance (DCR)” in comment and design item ID, should be “15.5mOhm Max” #1 (https://www.digikey.com/product-detail/en/bourns-inc/SRP6540-2R2M/SRP6540-2R2MCT-ND/3986035).
**L3**: Designator without problem.
**L4, L5**: Wrong ““DC Resistance (DCR)” in design item ID, should be “34.5mOhm” #1 (https://www.digikey.com/product-detail/en/w%C3%BCrth-elektronik/7443251600/732-4180-1-ND/3476496).
**L6**: Wrong inductance value in comment, should be 10uH and not 2.2uH #1  (https://www.digikey.com/product-detail/en/vishay-dale/IHLP3232DZER100M11/541-1346-1-ND/2616984).
**L7, L8, L9**: Designators without problem.
**P5**: Designators without problem.
**P8, RTD1, RTD2**: Wrong 3D model already noticed in #1, for this particular case it will stay this way.
** P11**: Designator without problem.
**PC104**: Designator without problem.
**Q1 and Q2**: Obsolete partnumber #2 (https://www.digikey.com/product-detail/en/vishay-siliconix/SI4010DY-T1-GE3/SI4010DY-T1-GE3TR-ND/6834463) new one should be (https://www.digikey.com/product-detail/en/vishay-siliconix/SI4166DY-T1-GE3/SI4166DY-T1-GE3CT-ND/3305374) this one is on Q8, Q9 and Q10.
**Q3, Q4, Q5, Q6**: Nothing wrong.

YanAzer, 2020-05-26:
**R65 and R66**: Have the partnumber (https://www.digikey.com/product-detail/en/vishay-beyschlag/MCT0603MC1002FP500/MCT0603-10-0K-MFCT-ND/3883103) while the rest of the similar resistors [**R1, R8, R18, R23, R31, R53, R58, R61**] have the partnumber (https://www.digikey.com/product-detail/en/vishay-beyschlag/MCT06030C1001FP500/MCT0603-1-00K-CFCT-ND/2607875).
**R2, R3, R9, R10, R15, R16, R20, R21, R46, R47, R50, R51, R55, R56, R93**: Designators without problem.
**R4, R11**: Designators without problem.
**R5, R12**: Designators without problem.
**R6, R13**: partnumber not available in Digi-key (https://br.mouser.com/ProductDetail/Yageo/AC0805FR-0711K5L?qs=%2Fha2pyFaduielNz2skYJtyTtm5XXYIxd2snA0hqJxrAhiacnDuqimw%3D%3D)
**R7, R14**: Designators without problem.
**R17, R22, R52, R57**: Designators without problem.
**R19, R24, R54**: Designators without problem.
**R25, R26**: Designators without problem.
**R27**: Designator without problem.
**R33, R42, R44**: Designators without problem.
**R34, R37, R39, R40, R43, R60**: Designators without problem.
**R35, R85**: Designators without problem.
**R36**: Designator without problem.
**R45, R70**: Designators without problem.
**59**:  Designator without problem.
**62**: Wrong power in description and design item ID, it should be 0,15W and not 1/8W = 0,125W 
**R63, R64**: Designators without problem.
**R67, R68, R72, R75, R79, R83**: Designators without problem.
**R69, R74, R81**: Designators without problem.
**R71, R73, R76, R82, R86**: Designators without problem.
**R_ESD**: Designator without problem.

YanAzer, 2020-05-27:
**Rsense3**: Designator without problem.
**RSense1**: Designator without problem.
** SBW, SW1, SW2**:
** SYSTEM_ON**: Wrong color of the led, should be orange (https://www.digikey.com/product-detail/en/lite-on-inc/LTST-C191KFKT/160-1445-1-ND/386832).
**U2, U14**: In Digi-key: “Not Recommended for New Design, minimums may apply. See Alternate Package or Substitute options” (https://www.digikey.com/product-detail/en/texas-instruments/TPS54540QDDARQ1/296-38301-1-ND/4967506).
**U3**: Not necessary the use of the SO8 Powed PAD footprint, the 8-SOIC package would be more compatible with the component (https://www.digikey.com/product-detail/en/texas-instruments/TPS5410QDRQ1/296-25229-1-ND/2174999).
**U4, U9**: Not necessary the use of the SO8 Powed PAD footprint, the 8-SOIC package would be more compatible with the components (https://www.digikey.com/product-detail/en/texas-instruments/TPS5420QDRQ1/296-23226-1-ND/1830821).
**U5**: Designator without problem.
**U6**: Designator without problem.
**U7**: Designator without problem.
**U10**: Designator without problem.
**U12**: Designator without problem.
**U13**: Designator without problem.
**U15, U16, U18, U20, U22, U25, U27**: Designators without problem.
**U17, U19, U21, U24, U26**: Designators without problem.
**U28**: Designator without problem.
**Y1**: Designator without problem.
**Y2**: Designator without problem.

YanAzer, 2020-07-07:
Found components for substitution and fixing obsolescence:

**C62, C77, C78**: [GRM188R71H104KA93D](https://www.digikey.com/product-detail/en/murata-electronics/GRM188R71H104KA93D/490-1519-1-ND/587854) **old** -> [GRM188R72A104KA35D](https://www.digikey.com/product-detail/en/murata-electronics/GRM188R72A104KA35D/490-3285-2-ND/702549) **new**

**Q1, Q2**: [SI4010DY-T1-GE3](https://www.digikey.com/product-detail/en/vishay-siliconix/SI4010DY-T1-GE3/SI4010DY-T1-GE3TR-ND/6834463) **old** -> [SI4166DY-T1-GE3](https://www.digikey.com/product-detail/en/vishay-siliconix/SI4166DY-T1-GE3/SI4166DY-T1-GE3CT-ND/3305374) **new**

closed 2020-09-01 by andrempmattos