spacelab-ufsc/eps2 #7: Hardware: Check if is necessary update the PC104 connector

andrempmattos, 2020-06-24:


YanAzer, 2020-07-13:
The current PC104 connector has the old partnumber (https://www.digikey.com/product-detail/en/samtec-inc/SSW-126-01-G-D/SAM1208-26-ND/1112032), same as previous EPS 1.0 module. By contrast the OBDH 2.0 uses the right PC104 partnumber (https://www.digikey.com/product-detail/en/samtec-inc/ESQ-126-39-G-D/SAM10462-ND/3601055). Changing only the partnumber should solve the bug.

The visual 3D models are simplayed below, the difference in height is noticeable. @andrempmattos 

[image]

andrempmattos, 2020-08-29:
Component updated! Using SSW-126-01-G-D

commit 447575f6 (2020-07-17): Hardware: libraries & pcb: fixing 3d model on pcb and schematic libraries and updating the pcb close #7
closed 2020-08-29 by andrempmattos