spacelab-ufsc/eps2 #22: Firmware: Define the firmware structure

mgm8, 2020-09-02:
Use the OBDH 2.0 firmware as reference: [https://github.com/spacelab-ufsc/obdh2/tree/master/firmware](https://github.com/spacelab-ufsc/obdh2/tree/master/firmware)

YanAzer, 2021-06-11:
1614dcb74de25a04e236f6bd0524266b4464d527
983c9b272aa215c3eca1ec4cb5141f4304884cec
8103af7ef73aa1c2c60d5b138dea1d5342ab83c2
c30d1199429d476f869f8fd2d69e88c9a979733e
d1122eeb8fef141d17667d814b9c07ad42ff267d
8e8f379845ea6e1491440d73c51a580082f8913b
69d73a21dfa7277e14bf0dda92dc7ef68b04a4a2
6875fb74cd00d63773a23efafd312870127c259d
9ce02371780ea19b0df6e440e6e20abb09361a59
d4289632fe89e984799a81b9fb73a5988d534adc
268e964468f74476983c1ab689bfd0bba2d4bc74
084006f83c1788c74193d096bd164aed4b212aea

andrempmattos, 2021-06-14:
@YanAzer, I think we finished this issue... so I will close it.

Your diagram shows that: 
https://github.com/spacelab-ufsc/eps2/blob/documentation/doc/figures/eps-system-layers.pdf

closed 2021-06-14 by andrempmattos