spacelab-ufsc/eps2 #44: Hardware: Increase the maximum current of the D17-D22 diodes

mgm8, 2020-09-30:
The current model has a maximum continuous current of 1 A. This value must be increased to at least 2 A (with 2 solar panels in parallel, the maximum nominal current can reach 1 A or more).

YanAzer, 2021-03-08:
Is this issue still relevant? If i am not mistaken the solar panels already include these blocking diodes, can you guys confirm @mgm8  @andrempmattos? Found an option [SMD22WS-TP](https://www.digikey.com/en/products/detail/micro-commercial-co/SMD22WS-TP/8554705?s=N4IgjCBcpgbFoDGUBmBDANgZwKYBoQB7KAbRABYAGSgJnIgF0CAHAFyhAGVWAnASwB2AcxABfArACcCEMkjps%2BIqQqVJAZg0gmINh279hYgpPjRZqTLgLFIZKpXUB2ABzaW7SF16CR4kJJg5DJyCtbKduCwTgCsksE6el4GvsbgNIEhloo2KlTkLuTSiZ7ehn4EcC5mSNnhtmR0GWBuJfo%2BRv5gki7S5qFWSg0glO66pSmdoqJAA) anyway.

mgm8, 2021-03-08:
Yes, most commercial solar panels already include blocking diodes. In the FM we will probably replace these diodes by 0R jumpers (or even remove it from the project). For now we can replace the part number, but we will have to think about removing it or not soon.

commit 270213fb (2021-06-15): hardware: libraries: Adding blocking diode of 2A to libraries
hardware: sources: Updating component in schematic and layout #45 close #44
closed 2021-06-15 by YanAzer