spacelab-ufsc/eps2 #55: Firmware: Define the system tasks

mgm8, 2020-11-13:


mgm8, 2020-11-18:
@YanAzer Preliminary list of tasks (more details in 53f2689):

* Startup (boot)
* Watchdog reset
* Periodic system reset
* Battery heater control
* Read sensors
* CSP server

> **TODO**: List all available data and sensors in the EPS module

YanAzer, 2020-11-20:
@mgm8 @andrempmattos some other tasks that may be missing:

• MPPT
• Energy level
• Send data Beacon
• Send data OBDH

Also a first list of available data (related to sensors) and sensors in the EPS module:

Data:
• Batteries temperature
• Battteries voltage
• Batteries current
• Solar panels temperature 
• Solar panel pair voltage 
• Individual solar panels current
• MCU internal temperature 
• Main bus voltage 

Sensors:
• RTDs
• Current sensors
• Voltage sensors
• Battery monitoring IC

andrempmattos, 2021-08-02:
Current tasks list:
- Startup (boot)
- Watchdog reset
- Periodic system reset
- Battery management
- Battery heater control
- MPPT algorithm
- Read sensors
- Parameter server
- _[CSP server]_ (probably won't be implemented)