spacelab-ufsc/eps2 #73: Firmware: Develop the heater device

mgm8, 2021-02-10:
Develop a device to handle the battery heater.

andrempmattos, 2021-08-04:
@joaoclaudioeb, check if we **must** use float variables! If so, which is the impact using the MSP430? Can we simplify the algorithm and use only intergers?

mgm8, 2021-08-04:
> @joaoclaudioeb, check if we **must** use float variables! If so, which is the impact using the MSP430? Can we simplify the algorithm and use only intergers?
> 
> For example, can we work with variables in other units, such as: (temperature * 100) or (temperature in kelvin) ?

If it's necessary to work float variables, using fixed-point arithmetic will save some resources: [https://www.best-microcontroller-projects.com/fixed-point.html](https://www.best-microcontroller-projects.com/fixed-point.html)

Working with other units (Kelvin instead of Celsius as example) is also a good approach.