spacelab-ufsc/eps2 #83: Documentation: List all variables from the sensors and peripherals of the EPS

mgm8, 2021-02-20:
Add a table with all possible variables that can be read from the sensors and peripherals of the EPS.

YanAzer, 2021-03-02:
Preliminary list:

**From current sensors on solar panels input (measured by EPS2 MCU):**
- Current solar panel pair side -Y
- Current solar panel pair side +Y
- Current solar panel pair side -X
- Current solar panel pair side +X
- Current solar panel -Z
- Current solar panel +Z

**From voltage sensors on solar panels input (measured by EPS2 MCU):**
- Voltage solar panels -Y and +X
- Voltage solar panels -X and +Z
- Voltage solar panels -Z and +Y
- Voltage output from boosters

**From current sensor on EPS/Beacon regulator (measured by EPS2 MCU):**
- Current from EPS2 circuitry and Beacon MCU;

**From voltage sensors on main power buss (measured by EPS2 MCU):**
- Voltage from main power buss (VBUS)

**From batteries monitor IC (measured by itself):**
- Voltage from positive side of the batteries (+VBAT)
- Voltage from common reference of the batteries (VBAT common)
- Batteries curret and average current
- Battery monitor IC temperature

**From RTDs on the batteries board (measured by external ADC IC):**
- Heaters temperature (RTDs)

**From EPS2 MCU (measured by itself):**
- Internal EPS2 MCU temperature

Note: some variables from the battery monitoring IC were not considered such as:

- Batteries monitor status register
- Batteries monitor cycle counter
- Batteries monitor remaining active absolute capacity register
- Batteries monitor remaining standby absolute capacity register
- Batteries monitor remaining active relative capacity register
- Batteries monitor remaining standby relative capacity register

@mgm8 @AugustoVassoler @vinepb

mgm8, 2021-03-04:
@YanAzer Suggestion: Add an ID number for each variable (16-bits?), and the length, in bytes, of each variable.

andrempmattos, 2021-08-02:
@mgm8 and @YanAzer, is this issue finished?