spacelab-ufsc/eps2 #86: Firmware: Delevop the struct for all available variables of EPS2

YanAzer, 2021-03-05:
Tasks will acess this struct to retrive data, including the OBDH2 task to retrieve the complete package of telemetry. This header file should be located under the firmware/app/structs subdirectory.

andrempmattos, 2021-08-02:
@YanAzer and @mgm8, I added a scheme to abstract the structure R/W accesses. The implementation now is simple and straightforward, but we can improve from that by adding redundancies for the buffer (i.e., 3 copies) or CRC for blocks of data or even creating a restrictive policy to avoid errouneos/simultaneous accesses.

andrempmattos, 2021-08-03:
I suggest the change:

MPPT 1  **--> MPPT 0**
MPPT 2 **--> MPPT 1**
MPPT 3 **--> MPPT 2**

andrempmattos, 2021-08-04:
I suggest the following inclusion:

**HEATER_SET_POINT** (For manual mode)