spacelab-ufsc/eps2 #103: Hardware: Evaluate component values used in the current sensor circuit

YanAzer, 2021-04-12:
This time the EPS will be receiving twice the current because of two solar panels in parallel, evaluate the values used and see what it needs to be changed.

YanAzer, 2021-05-05:
Prelimary evaluation resulted in the change of both Rsense and Rout to maintain the same measurement caracteristics from previous EPS project:
- Rsense = 20mΩ
- Rout = 4kΩ

For this values and with a fixed Vout = 3V (different from the previous project that was 2.5V) and considering a current load of 1.5A (worst case for this situation) with the formulas from the MAX9934 datasheet results:

- Iout = 750uA (same as EPS 1.0)
- Vsense = 30mV (same as EPS 1.0)
- Psense = 45mW (EPS 1.0 1.8mW considering a current load of only 600mA)

Values of Rout below 4kΩ would result in a greater current output of  750uA from the MAX9934, it is not certain the maximum limits for this parameter, but this value is the same constraint used in the EPS 1.0 project.

YanAzer, 2021-05-12:
@mgm8 @andrempmattos my calculations seem right? If not, well, it will be just boring desoldering these resistors afterwords...

andrempmattos, 2021-06-06:
@YanAzer, I think yout assumption that Vout might be 3V need a review. The VREF+ uses the external voltage reference circuit that generates 2V5 in the output. I think the absolute maximum rate of the ADC input is the AVCC supply (3V3), but to have readable value the max output in the ADC pins should be 2V5. Please, check that before the next release.

In page 48 you will find more info: https://www.ti.com/lit/ds/symlink/msp430f6659.pdf?ts=1622990330170&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FMSP430F6659

Voltage reference circuit: https://www.ti.com/product/REF5025A-Q1

YanAzer, 2021-06-06:
@andrempmattos thanks for the feedback, you are correct, redoing the previous calculations with the same parameters and Vout = 2.5V we obtain: Rsense = 20.2mΩ and Rout = 3.3kΩ. I will be improving this part for the documentation too.

commit 587cc8cf (2021-05-16): hardware: sources: board_layout: Adding jumper close #38 and updating shunt and output resistor close #103 on board layout
closed 2021-06-14 by andrempmattos