spacelab-ufsc/eps2 #140: Firmware: Develop a UART with interruption for handling TTC communication

andrempmattos, 2021-08-01:


mgm8, 2021-08-02:
Not only the TTC, but any device connected to the serial interface (UART port).

andrempmattos, 2021-08-02:
Yes, I developed a generic UART driver for communication with interruptions. I just need to improve the ISR routine part that curretly only works for USCI_A0 and inside it calls the param server task scheme (very specific to the TTC interface). If you have a better suggestion concerning this last part... otherwise I think for now I will let as it is, i. e., ISR should be added/modified for another use case.

Also, just an observation, in order to implement the isr inner routine more generically, I used the UART break char feature to flag the end of a communication, otherwise the EPS would not know the end of a package since the UART only has the RX interrupt flag for this purpose (different from I2C that has a stop bit flag).