spacelab-ufsc/eps2 #201: License: Consider adding a hardware-specific license to the hardware components

DeflateAwning, 2024-10-04:
Would love to see a hardware specific license (e.g., Solderpad, CERN Open Hardware, etc.) for the hardware-specific parts! Open hardware is a great movement, and I'm very excited to explore this project more, but would love to see it licensed in a way that more clearly represents the allowable usage of its parts

joaoclaudioeb, 2024-10-04:
Hello, @DeflateAwning, how are you? I hope you're doing well! We're really happy to know you're interested in the project. We fully agree with the idea of using a hardware-specific license for this (and other) projects. In fact, this project is already under the CERN Open Hardware License, version 2. This project is divided into three main areas: documentation, hardware design, and firmware design. As such, we have different licenses for each area. Unfortunately, this isn't clearly stated in the project. Therefore, I just wanted to let you know that:

1) If you're interested in using it, the license to be considered is the CERN OHL, version 2.

2) We will make this clearer in the project. Today, there is a brief description in the hardware folder’s README.md. I will make the licenses more clearly visible in each area and add more detailed information in the README.md regarding our philosophy on licensing our projects.

If you have any more suggestions or questions, feel free to share them. Thank you very much for your interest.

DeflateAwning, 2024-10-04:
Awesome! In the README of this project overall, it'd be awesome if you mentioned the hardware license!

closed 2025-09-13 by mgm8